adafruit/Adafruit_Floppy #1: capture_track waiting for index pulse

Niteto, 2022-01-01:
Capturing a track from index pulse to next index pulse will run into trouble when reading disks that are not index aligned, e.g. Amiga Disks. A sector might start before an the index and end after the index, flux transitions might be lost when the capture does not overlap.

ladyada, 2022-01-01:
yah right now we're using 3.5" floppy drive only since we have those :)
what should we do in that case, just capture for n milliseconds? or 1.25 'rotations'?

ladyada, 2022-01-02:
i think its fixed with https://github.com/adafruit/Adafruit_Floppy/commit/c84b536c647cd48e65ef6fdf1651c18814ee0373 - we read through the index pulse so we get an 'overlap' of readings

closed 2022-01-02 by ladyada