adafruit/Adafruit_Floppy #2: density pin

Niteto, 2022-01-01:
The density pin is used differently depending on the manufacturer, some use it as an output to signal the media type (mostly teac drives) some use it as an input to switch the rpm (300/360) when using HD media (e.g. Sony 920), and at some it is NC.
Detecting the media type can only be done by counting the transitions when reading a track. Or when writing a track by firstly trying to write a HD track and reading it back while counting the transitions, a DD disk will return much less transitions than written.
And another problem is that all HD drives change the sensitivity/write strength of the head depending on the sensor for the HD hole of a disk, so that a DD image written to a HD disk will fail. Covering the HD hole will circumvent this issue.

ladyada, 2022-01-01:
yeah we are not using it yet!

jepler, 2024-11-21:
The GreaseWeazle-compatible firmware supports the "set pin" functionality of the GreaseWeazle protocol, so that the desired state of Pin 2 can be selected at runtime.

closed 2024-11-21 by jepler