adafruit/Adafruit_Floppy #3: Drive emulation

jpnielsen, 2022-01-03:
Thank you for making this.

Will it emulate a floppy drive too?

I have an old Korg N364 synthesizer with a broken drive. Would be nice to mount a Pico instead, perhaps with extra flash.

ladyada, 2022-01-03:
we don't have any plans for that *yet* but it ought to be possible to write the inverse of all the current functions! it would have to deal with the level shifting more seriously since there's a greater chance of having a 5V signal come in.

jepler, 2024-11-21:
As of release 0.5.0 on the unreleased Adafruit Floppsy, read-only FM and MFM drive emulation is supported. This does not work with the Adafruit Floppy Featherwing because the direction of the data signals cannot be switched. It could work on homebrewed hardware.

closed 2024-11-21 by jepler