adafruit/Adafruit_Floppy #5: Has gw been modified for this project ?

jluke1, 2022-01-13:
When using your arduino pico code I get the following error.
Arduino 1.8.16,  macOS 10.15.7,  Greaseweazle v0.31

$ ./gw read --device /dev/cu.usbmodem14101 xxx
ERROR: Greaseweazle firmware v1.0 is unsupported
To perform an Update:
 - Run "gw update" to install firmware v0.31

Brilliant piece of work (again).

Thank you
John Luke

jluke1, 2022-01-13:
Does work if you change to
#define GW_FIRMVER_MAJOR 0
#define GW_FIRMVER_MINOR 31
 in greaseweazle.ino

John Luke

jluke1, 2022-01-13:
That should now become
#define GW_FIRMVER_MINOR 37

jluke1, 2022-01-14:
Answer to my own question. Done in greasweazle 0.36.

closed 2022-01-14 by jluke1

ladyada, 2022-01-14:
lol i was wondering what was up! soudns like its all working then :)

jluke1, 2022-01-14:
Well I have not tried plugging it in to a real drive. (Sigh somewhere in a box in the garage).
Do have a 3.5” usb drive but was not able to work out how to open it. May try plugging it
into a Gotek out of curiosity.

Yes gw 0.31 does insist on firmware 0.31. You might want to add a warning to the README
that greaseweazle >= 0.36 is required. Also is tinyUSB recommended for pico ?

Mildly irritating as have recently forked out $70 for a greaseweazle f7 lightning.

Thank you for the great work.

John Luke

ladyada, 2022-01-14:
hiya, you should be able to use Pico with the pinout indicated. 

please support greaseweazle development with your purchase! its important that the software writer can sell hardware to pay for all the coding they do :)
if you'd like, please submit a PR to the readme for any changes/stuff you encounter