adafruit/Adafruit_Floppy #13: multiple revolutions

jepler, 2022-02-11:
It appears that specialty operations, such as using g64conv to convert the second side of a flippy disk from a two-sided .scp of the disk taken from the top side, may require operation similar to the genuine greaseweazel hardware: capture multiple revolutions without gaps.

the current implementation in this repo simply captures 3 different copies of the same track, from one falling index pulse to the next.

Making a change like this has a big impact on the structure of the code. A ping-pong buffer system could be adopted, so that flux is entering buffer A and then buffer B; once buffer A is filled, it can be transmitted over USB. Probably rp2040 is forced to go interrupt based to enable this, which we want to do anyway.

I don't have any proof that this is why my flippy-floppy dump didn't work; there are also g64conv bugs—with the geniune greaseweazel scp file, the directory data which should be on track 18 is on track 14 of the resulting g64 file! but it is intact. (track 18 sectors 0 and 1).  By contrast, my rips with feather rp2040 to scp do not have track 18 sector 0, but do have track 18 sector 1 (again, transposed incorrectly to track 14 by g64conv). I know the flip side reads correctly because it happens to have two index windows and can be flipped in a standard PC 5.25 drive (teac).

ladyada, 2022-02-11:
we could probably do 2 revs continguously - bump the buffer to 225K on the RP2040 and 164K on the '51. are you using fluxengine or gw?

jepler, 2022-02-11:
the workflow for g64 that I have is to produce a .scp file with greaseweazel and then a .g64 file with g64conv.

closed 2024-11-21 by jepler