adafruit/Adafruit_Floppy #22: Feature Request: Roland D-10/20/50 Support

NoodleSploder, 2022-12-01:
There are people in the world that would like to have this available to pull their arrangements off of Roland's proprietary floppies. I have the schematics and pin outs. It is similar to the Apple II interfaces. I will contribute as much as possible if this is feasible.

trondl909, 2023-06-03:
We're trying to solve this problem over at VCFed.org:
https://forum.vcfed.org/index.php?threads/roland-d-20-decoding-the-mysterious-floppy-format.1243226/

Please, come over and help, especially if you know Arduino coding and stepper motor controlling.

The situation so far is that it looks like an Apple Disk II interface with a Brother240-ish format (GCR).
I have rudimentary logic analyzer logs of the stepper motor behavior during reading and formatting.
I'm trying to use an Arduino Mega 2560 to control the stepper and dump the content with the D-20 floppy drive itself, ideally at flux level.

jepler, 2024-11-21:
Best of luck with your project! This is beyond the scope of what we want to provide in the library or its examples, but it's possible that this library's code for reading & writing floppy flux could be useful to you.

closed 2024-11-21 by jepler