adafruit/Adafruit_Floppy #24: Greaseweazle stuck on read

AidenShaw2020, 2023-01-06:
Hi, i have issue with Adafruit_Floppy (Raspberry Pi Pico RP2040 (philhower core)), using Greaseweazle.
When i try to read ibm.1440 (gw read test.img --device=com14 --format=ibm.1440), floppy drive goes to green and spin, but i dont get any data and progress.

[image]

I use last release (0.2.0) of Adafruit_Floppy and Greaseweazle 1.4.

Can you help me please?
Thanks

AidenShaw2020, 2023-01-06:
I also try floppy_capture_track_test, and last version on GIT, but without luck.
Wirings is also OK, and without shortages.
[image]

AidenShaw2020, 2023-01-06:
Its seems, as RP2040 have insufficient voltage to trigger head stepper motor controller to work.

AidenShaw2020, 2023-01-06:
Addendum: I tryied three different FDD, and state is same.

AidenShaw2020, 2023-01-06:
Addendum, adddendum: If i use 0.1.0 (floppy_capture_track_test), floppy it seems work: 
[image]

ca0abinary, 2023-01-07:
Same problem here. Too late to track down the regression, but this got me up and running:
```sh
git checkout b650a78743eca695132833b026f2771f45e1cdb0
echo 'diff --git a/src/Adafruit_Floppy.h b/src/Adafruit_Floppy.h
index 85993e1..9d44416 100644
--- a/src/Adafruit_Floppy.h
+++ b/src/Adafruit_Floppy.h
@@ -128,7 +128,7 @@ public:
                       adafruit_floppy_disk_t format = IBMPC1440K);
 
   bool begin(void);
-  bool end(void);
+  void end(void);
 
   uint32_t size(void);
   int32_t readTrack(uint8_t track, bool head);
diff --git a/src/Adafruit_MFM_Floppy.cpp b/src/Adafruit_MFM_Floppy.cpp
index 174cebd..e10f407 100644
--- a/src/Adafruit_MFM_Floppy.cpp
+++ b/src/Adafruit_MFM_Floppy.cpp
@@ -53,10 +53,10 @@ bool Adafruit_MFM_Floppy::begin(void) {
     @returns True always
 */
 /**************************************************************************/
-bool Adafruit_MFM_Floppy::end(void) {
+void Adafruit_MFM_Floppy::end(void) {
   _floppy->spin_motor(false);
   _floppy->select(false);
-  return true;
+  // return true;
 }
 
 /**************************************************************************/
' | patch -p1
```

It just basically reverts to an earlier version of the code and patches a return type change. Afterwards I uploaded and was able to get a flux file from a disk without any issues.

AidenShaw2020, 2023-01-07:
Hello ca0abinary, thanks for help! Now head is moving, and trying to read, but without success (GW): 
[image]
Flux i cannot try (device is not recongnised).
Thanks.

AidenShaw2020, 2023-01-07:
Write crash at fatal error:
[image]

ca0abinary, 2023-01-08:
Hey @AidenShaw2020, I've struggled with the same issues. One disk would behave like yours for reading, another read fine. I've found using `--raw` and saving as a `.sfd` file can help but I'm a complete novice at using Greaseweazle still; there are likely other parameters that can help. As far as writing is concerned that doesn't work at all for me yet, still haven't gotten time to really dig into the code or setup a debugging environment. You're one step ahead of me with the ability to capture serial1 output.

AidenShaw2020, 2023-01-08:
Hi @ca0abinary. For serial debug, you can use CH340 USB to TTL UART (for. example: https://www.laskakit.cz/prevodnik-usb-ttl-uart--ch340), and connect it RX on USB TTL to GPIO 0 (and GND to GND of course), and baudrate is: 115200.

AidenShaw2020, 2023-01-08:
I have breakthrough :D !!!
If i add 6,8k resistor between IDC 30pin (RDATA / READ), and GPIO12 on RP2040, GW is working OK.
[image]
[image]

Also write it back, and read on regular PC drive is OK:
[image]
[image]

AidenShaw2020, 2023-01-08:
But now i have problem with Flux, its still not working :(
[image]

ca0abinary, 2023-01-08:
Does the most recent code work with the resister? I remember reading about a fix for fluxengine.

AidenShaw2020, 2023-01-08:
With new code, is state same as begin of this issue (doesnt not start read).

ca0abinary, 2023-01-08:
Thanks for testing. For me the 6.8k resistor inline with IDC30 doesn't help
```text
IDC30 --- /\/\/ --- GPIO12
```
I've determined the regression started at commit d6dc0fe043ae2aeaf876de4611cae175b003634a. Will look into it more later today.

AidenShaw2020, 2023-01-08:
Maybe try it with different value (between 4.7 - 10k).

charlysan, 2024-02-26:
Hey there,

I've tried release `0.2` and `0.3` with the same results on my Pico. I don't get any movement, spin or green light on my floppy drives. I've checked that the drives works using a PC, and tried with two different working drives. I have also tried adding a resistor from IDC30 to GPIO12.

I'm using the following pinout:
```
#define DENSITY_PIN 2 // IDC 2
#define INDEX_PIN 3   // IDC 8
#define SELECT_PIN 4  // IDC 12
#define MOTOR_PIN 5   // IDC 16
#define DIR_PIN 6     // IDC 18
#define STEP_PIN 7    // IDC 20
#define WRDATA_PIN 8  // IDC 22 (not used during read)
#define WRGATE_PIN 9  // IDC 24 (not used during read)
#define TRK0_PIN 10   // IDC 26
#define PROT_PIN 11   // IDC 28
#define READ_PIN 12   // IDC 30
#define SIDE_PIN 13   // IDC 32
#define READY_PIN 14  // IDC 34
```

```
bin/gw info --device /dev/cu.usbmodem21101
Host Tools: 1.16.2
Device:
  Port:     /dev/cu.usbmodem21101
  Model:    Adafruit Floppy Generic
  Firmware: 1.0
  Serial:   E6611CB697768F25
  USB:      Full Speed (12 Mbit/s)
```

Tried to read using the following command with these results:
```
 bin/gw read --device /dev/cu.usbmodem21101  --format ibm.1440 --retries 1 test.img
Reading c=0-79:h=0-1 revs=2
Format ibm.1440
Command Failed: Motor: No Index
```

Serial console shows this:
```
Got command 0x00 of length 3
Get info
Got command 0x05 of length 4
Get params
Got command 0x0e of length 3
setbustype() floppy = 0x200023a8 result=1
                                         Set bus type 1 -> 1
Got command 0x0c of length 3
Select drive 0
Reply buffer = 12 0
                   Got command 0x06 of length 4
Turn motor 0 on
Waiting for index pulse...Didn't find an index pulse!
Timed out waiting for command, resetting motor
goto track 0
Going to track 0
Could not find track 0
stop motor
deselect
motor reset
Got command 0x06 of length 4
Turn motor 0 off
Got command 0x0d of length 2
Deselect drive
Timed out waiting for command, resetting motor
goto track 0
Going to track 0
Could not find track 0
stop motor
deselect
motor reset
Timed out waiting for command, resetting motor
goto track 0
Going to track 0
Could not find track 0
stop motor
deselect
motor reset
Timed out waiting for command, resetting motor
goto track 0
Going to track 0
```

Any suggestion?

charlysan, 2024-02-27:
My bad. I was using the wrong platform. I had to choose `pico_rp2040_tinyusb`. I could successfully read a 1.44MB IBM format floppy disk from the Pico.

jepler, 2024-12-02:
Closing this as both @charlysan and @AidenShaw2020 have reported success.

closed 2024-12-02 by jepler