adafruit/Adafruit_Floppy #25: Request for wiring a apple ii 5.25 20 Pin to RP2040 pico/feather

1st-elektron, 2023-01-24:
Thank you for opening an issue on an Adafruit Arduino library repository.  To
improve the speed of resolution please review the following guidelines and
common troubleshooting steps below before creating the issue:

- **Do not use GitHub issues for troubleshooting projects and issues.**  Instead use
  the forums at http://forums.adafruit.com to ask questions and troubleshoot why
  something isn't working as expected.  In many cases the problem is a common issue
  that you will more quickly receive help from the forum community.  GitHub issues
  are meant for known defects in the code.  If you don't know if there is a defect
  in the code then start with troubleshooting on the forum first.

- **If following a tutorial or guide be sure you didn't miss a step.** Carefully
  check all of the steps and commands to run have been followed.  Consult the
  forum if you're unsure or have questions about steps in a guide/tutorial.

- **For Arduino projects check these very common issues to ensure they don't apply**:

  - For uploading sketches or communicating with the board make sure you're using
    a **USB data cable** and **not** a **USB charge-only cable**.  It is sometimes
    very hard to tell the difference between a data and charge cable!  Try using the
    cable with other devices or swapping to another cable to confirm it is not
    the problem.

  - **Be sure you are supplying adequate power to the board.**  Check the specs of
    your board and plug in an external power supply.  In many cases just
    plugging a board into your computer is not enough to power it and other
    peripherals.

  - **Double check all soldering joints and connections.**  Flakey connections
    cause many mysterious problems.  See the [guide to excellent soldering](https://learn.adafruit.com/adafruit-guide-excellent-soldering/tools) for examples of good solder joints.

  - **Ensure you are using an official Arduino or Adafruit board.** We can't
    guarantee a clone board will have the same functionality and work as expected
    with this code and don't support them.

If you're sure this issue is a defect in the code and checked the steps above
please fill in the following fields to provide enough troubleshooting information.
You may delete the guideline and text above to just leave the following details:

- Arduino board:  **INSERT ARDUINO BOARD NAME/TYPE HERE**

- Arduino IDE version (found in Arduino -> About Arduino menu):  **INSERT ARDUINO
  VERSION HERE**

- List the steps to reproduce the problem below (if possible attach a sketch or
  copy the sketch code in too): **LIST REPRO STEPS BELOW**

1st-elektron, 2023-05-12:
Please can someone answer my question about:
wiring a apple ii 5.25 20 Pin to RP2040 pico/feather

Thank you

montyleib, 2024-07-11:
The short answer is that Apple ][ drives with 20-pin connectors are incompatible with Adafruit-floppy and Greaseweazle.

The longer answer is that the control signals required to operate an Apple ][ drive are not the same as a standard PC drive, and the software does not include routines to create them.

Issue should be closed, unless someone wants to make it a feature request...

jepler, 2024-11-21:
The current prototype of the Adafruit floppsy has an Apple Disk ][ connector. Since this is unreleased hardware, the full schematic isn't available yet. However, here's a screen grab of the relevant bit:
[image]
In addition, a drive modification is used to provide a once-per-revolution signal.

closed 2024-11-21 by jepler

1st-elektron, 2024-11-21:
Thank you Jeff.

montyleib, 2024-11-28:
Thank you, Jeff!

gilmod714, 2024-12-22:
what about Trk0 signal ? .. where would it come from?!

jepler, 2024-12-23:
The drive does not have a trk0 signal. This kind of drive finds its home position by simply stepping outwards enough times to ensure it's reached the outermost track. 

This is handled in the `Adafruit_Apple2Floppy` class:
```c
bool Adafruit_Apple2Floppy::goto_track(uint8_t track_num) {
  if (_quartertrack == -1) {
    _quartertrack = 160;
    goto_quartertrack(0);
```
If the `_quartertrack` member variable is -1, then 160 quarter-steps inwards are performed. In my drive, this was always enough to end up at track 0. However, this is from a sample of just 1 drive, so it might be that some drives can reach further, beyond quarter track 160.

gilmod714, 2024-12-23:
Awesome! thx Jepler