adafruit/Adafruit_Floppy #33: Info on pi pico pinout and compiling.

longboarder241, 2024-06-09:
I saw the video on the playlist describing how this firmware can be used with the pi pico however setting this up with this device is very minimal.  
(https://youtu.be/hcgtolaM_6g)
The wiring to the pico is quite descriptive," start at GP2 and connect all the pins thru to GP14 in a row, skipping over the ground pads (dont forget to connect one of the grounds of course)" however I cannot properly figure out which pins interface directly with the floppy disk drive.
Perhaps the precompiled firmware for the feather rp2040 will be cross compatible with my pico but I don't know for sure until I have the ability to wire it to my disk drive.  If that is not the case it'd be helpful to know how to set that up.

Regards.

mbulandra, 2024-10-06:
The wiring is clearly defined in greaseweazle.ino file, starting at line 31. The numbers in code are GPIO numbers. Remember to connect some GND pins as well, and READ signal needs to be pulled up through a ~5kΩ resitor.

jepler, 2024-11-21:
Thanks for answering @mbulandra !

closed 2024-11-21 by jepler