adafruit/Adafruit_Floppy #41: Always failing to find Track 0

criscrafter, 2024-12-04:
I am using a Pi Pico and compiling in VS Code through PlatformIO.

I always get an error when it is trying to find Track 0.
I had an issue before where it wouldn't find the index pulse, but I did fix that with a pull resistor on the Index pin.
Using pull-up resistors however didnt seem to fix anything on the Track 0 pin.
I tried 5 differen floppy drives, all with the same outcome.

I need to archive about 500 floppys soon, so I would really love to get this working.

Any Ideas?

Greetings, Friedrich.

mrenters, 2025-02-01:
What type of drive are you using? I have a Shugart 400 5.25" drive and was having similar symptoms. There is a jumper (MH) on the SA400 drive that allows you to select whether the head loads when the drive is selected or when the motor is on. My drive was set to load the head when the motor is on and the track 0 signal would only be active when the motor is on. Removing it causes the head to load when the drive select pin is activated and the track 0 signal also becomes active. With the MH jumper in place, the drive wouldn't step the head if the motor wasn't on. It looks like newer drives load the heads when a floppy is inserted and thus don't have this issue.

My drive needed a longer step delay to reliably step. I set it to 12ms and it worked fine this. That change was just committed to the main branch.

montyleib, 2025-04-06:
@criscrafter Friedrich, do you have any additional details, failed attempts, etc?  What is the error you see?

I'm hoping that your "batch" routine can set aside any "rips" with read errors so that you can manually work on them...  archiving is much more demanding than "load and walk away".

Monty