adafruit/Adafruit_Floppy #43: No Erase in gw ?!

gilmod714, 2024-12-21:
I built the pi pico GW version and it works OK for read/write with 4.7k resistor on read line  (it's a bit slow, but it works) , however, I noticed that, the "erase" disk function does not work, or not supported ?!

Great work by the way ..

Thank you

criscrafter, 2024-12-22:
Are you pulling the pin up to 5V or 3.3V?
Sadly I don't have a solution for you, but I can't get mine to work...

gilmod714, 2024-12-22:
I'm using 3.3v pull .. but I don't think it's that .. it seems to me the "erase" function isn't included in the sketch! .. the read/write works ok though .. but much slower than the stm32 for sure.. I'm using RP2040 zero/and also tried it on pi pico ..

[image]

[image]

[image]

jepler, 2024-12-23:
EraseFlux (command #17) is not presently supported by our gw-compatible firmware. 

In the official gw firmware, it appears this mode works simply by turning on the write gate and leaving it on for a specified time:
```c

/*
 * ERASE PATH
 */

static uint8_t floppy_erase_prep(const struct gw_erase_flux *ef)
{
    op_delay_wait(DELAY_write);

    if (get_wrprot() == LOW)
        return ACK_WRPROT;

    write_pin(wgate, TRUE);

    floppy_state = ST_erase_flux;
    flux_op.status = ACK_OKAY;
    flux_op.end = time_now() + time_from_samples(ef->ticks);

    return ACK_OKAY;
}

static void floppy_erase(void)
{
    if (time_since(flux_op.end) < 0)
        return;

    floppy_flux_end();

    /* ACK with Status byte. */
    u_buf[0] = flux_op.status;
    floppy_state = ST_command_wait;
    floppy_end_command(u_buf, 1);
}```

I've marked this "help wanted" because I'd be thrilled if someone contributed the implementation.

gilmod714, 2024-12-23:
It has to erase track by track i think, perhaps not just a delay period..

montyleib, 2025-04-06:
I believe I've succeeded in adding GW_CMD_ERASE but have no way to test it.  We're in the process of selling our home of 32 years, and we've been packing away all of the non-essentials in preparation for a move.  Needless to say, all of my "old iron" is now boxed up.  :(

@jepler   Jeff - if you can take a look at my fork [https://github.com/montyleib/Adafruit_Floppy/blob/main/examples/greaseweazle/greaseweazle.ino](url) lines 337 to 373, I'd love to get your feedback before I write a commit.

gilmod714, 2025-04-06:
Awesome work @montyleib .. Thank you for you efforts and wishing you a smooth and easy move :)

gilmod714, 2025-08-01:
Hello @montyleib ,

I appreciate your work .. sorry I didn't see your previous link before now .. however, the link above takes me to a blank page though .. where can I find your .ino ?! .. so i can test it out and report back

Thank you!

[image]

montyleib, 2025-08-01:
Strange, it does the same for me.  Mine on  GitHub is https://github.com/montyleib/Adafruit_Floppy/blob/main/examples/greaseweazle/greaseweazle.ino but I suspect it won't run...  it appears that I failed to re-base some variables copied over from GW.

montyleib, 2025-08-02:
greaseweazle.ino only has 729 lines...  and nothing even remotely close to the code in the sketch window.

gilmod714, 2025-08-02:
oops! .. not sure where i got that from! .. here's the correct one - no compile

[image]

ladyada, 2026-10-12:
thanks, fixed & added!

closed 2026-10-12 by ladyada