adafruit/Adafruit_Floppy #44: Possible to update GW firmware to .27 or higher ?!

gilmod714, 2025-01-01:
Hello,

would it be possible to update the greaseweazle FW to .27 or higher ? .. the current .22 is too old that the example is based on.. "greaseweazle.ino"

the reason being .. I'm trying to connect the floppy to WinUAE as amiga drive plugin , and it complains about the old GW fw..

[image]

jepler, 2025-01-23:
We'd be happy to see support for newer GW messages added, and when they are added we can bump the compatibility firmware version number.

gilmod714, 2025-01-23:
Awesome !!!

montyleib, 2025-01-31:
Jeff, I think gilmod714 is asking for the Feather Greaseweazle firmware to be updated, which isn't in this repo.  Where would one go to help out?

gilmod714, 2025-01-31:
YES .. exactly .. Thanks Montyleib

montyleib, 2025-02-05:
Thanks, Jeff!

I had only found it in keirf's firmware repository, and assumed his is behind this one.

montyleib, 2025-02-05:
I see 21 (assuming .21) in the code.  No one has kept up with keirf, although Jeff has sure been very involved due to his work on Apple floppy code.

I've tried to compare this with keirf's current firmware, and it's done little more than give me headaches.  Assuming greaseweazle.ino is modelled after .21, I'll look backward in keirf's code and try to go from there.

No promises, though...  :)

gilmod714, 2025-02-06:
Any efforts to bring up the current firmware above .27 would be very useful I think :) .. 21 is really-really old FW! 

Thanks for your efforts in advance montyleib :)

montyleib, 2025-02-08:
@ladyada set the firmware version in greaseweazle.ino as 1.0 (lines 91 and 92), which is what the Greaseweazle host tools reports when "gw info" is run.  Seems curious that WinUAE reports 'too old' and 'need .27 or greater'.  I'm suspicious that the device model that is reported (Adafruit Floppy Generic) is the real issue.

I'm currently comparing the blue pill C code between versions, I still need to wrap my head around a) figuring out the translation C->Python and b) make the changes in Python to match the C from .22... it's making more sense daily.  :)

@gilmod714 : are you certain that .22 is what WinUAE thinks greaseweazle.ino is?  I have keirf's source from .22 to .31 (which became 1.0)

from keirf's Release Notes:
-----------------------------------------------
** v1.0 - 26 December 2021
 - Firmware is now released separately from the host tools
 - No firmware changes since v0.31 (apart from the version bump)
-----------------------------------------------

ladyada, 2025-02-09:
@jepler ^ want to take a look in the next few weeks?

gilmod714, 2025-02-09:
@montyleib .. yes, I believe WINUAE reports it as .22 on my RP2040 (I forgot where I saw that exactly, but yeah almost certain that's what it reported, when I was tinkering with it - trying to make it work as a disk plugin)

Thank you for your hard work :)

montyleib, 2025-04-06:
@gilmod714 .. ~~it just occurred to me that WINUAE is sending CMD_MAX and deciding whether the firmware is capable based on the returned report.  If 16 is returned, for example, it knows that the firmware version is below when keirf added ERASE in 2020.~~  Nope, greaseweazle.ino reports CMD_MAX as 26, so I'm guessing that WINUAE is sending one of the missing commands and getting a BAD_CMD response.  It may well be 11 SWITCH_FW_MODE, used to swap between update firmware and run modes, but could be 21 CMD_TEST_MODE or 22 CMD_NOCLICK_STEP.

The current version of the original Greaseweazle firmware has 23 commands available, where 10 was skipped and 18 thru 22 appear to be for testing.  greaseweazle.ino has 21, where I'm working on 17 and Jeff has implemented 18 thru 20 while working on his Apple drive code, as well as adding 26 specifically to check the status of the track 0 pin.  (@jepler may I suggest a slight rewrite to use GW_CMD_GETPIN TRK0_PIN instead, so that it follows whatever pin assignment is in use?)

Monty

montyleib, 2025-04-07:
@ladyada - you let him retire?  :)  Has anyone stepped in to continue his Apple floppy interfacing work?

With that in mind, I'll continue work on catching up with keirf's code.  I located my 2040 and floppy featherwing in the stack of boxes, just have to dig a little more to pull a 5.25 and 3.5 back out so I can test my code before submitting.

I knew I should have held off packing them... or at least made sure they were at the top of the stack...  :)

gilmod714, 2025-04-07:
Jepler retired ?!  .. I was looking forward to the apple drive completed code .. I'm glad I have switched my plans and converted my Apple drive into a 1541 instead 

https://www.lemon64.com/forum/viewtopic.php?t=86143

@montyleib .. thank you for your hard work :)

ladyada, 2025-08-15:
hiya PR's are welcome as long as you document what ya did manage to test. we'll be getting back to floppsy soon now that fruit jam is shipping

ladyada, 2026-10-13:
fixed!

gilmod714, 2026-10-13:
Awesome work, thank you @ladyada , is there a new "release" or just fixed .ino in the repo ?!