adafruit/Adafruit_Floppy #46: mfm_emu not compile and improvement req

aotta, 2025-01-22:
I like very much the MFM emu example for the possibility to emulate a floppy with a simple Pico, it opens a world of possibility.
But it doesn't compile for me as is , i got this error and also cur_format is not declared in line 420 and 452:
```
C:\Users\Andrea\Documents\Retrocomputer\Sharp\X68000\X68KFDPI2\Adafruit_Floppy-main\examples\mfm_emu\mfm_emu.ino:347:20: error: 'cur_format' was not declared in this scope
  347 |       .n_sectors = cur_format->sectors,
      |                    ^~~~~~~~~~
```

i made some changes and got a quite working versions for emulate a Sharp X68000 5,1/4 floppy 1200k, the pico start sending data on read_pin when the PC starts, but the Sharp doesn't get the right data.

It seems the example is tailored for the Xerox 820, is possible to have some more info and comments into the code to let users to customize it?
Thank you in advance and my congrats for the project!

jepler, 2025-01-23:
I updated the CI and confirmed that the current main branch version of the example does compile.

Yes, it would be a fine contribution to add support for more FM/MFM image types, or the types themselves.

Eventually, Adafruit will document this better when our floppy flux copier / emulator HW actually gets to release. For now, we've been more interested in just verifying that the functionality we want to be sure works, does actually work. But it's not polished for end users yet.

aotta, 2025-02-01:
It's for sure interesting that this demo program works, but it has a lot of hardcoded lines that seems tailored for Xerox820drive or in any case 300rpm ones (i.e. lines 94, 150,151).
In my test i also added a line for supporting 1232KB drives in the floppy_format_info_t:
  {77,  8, 2, 1000, 166667, 3, false}, // 5.25" 1232kB, 360RPM
  
but the flux sent to my SharpX68000 is not recognized by the PC, i think it need the HFE format of the image file, since it's the only format that works in others phisical floppy emulator too.

So, a very appreciated improvements in mfm_emu would be adding the support for reading also HFE files other than IMG ones, and using the original MFM format instead of the encoded_mfm provided by the Adafruit floppy library.  Thank you all for supporting this great and unique library, and waiting for the Adafruit Floppsy!!

ladyada, 2026-10-12:
fixed!

closed 2026-10-12 by ladyada