adafruit/Adafruit_Floppy #48: Cannot compile forked repo

eparadis, 2025-01-23:
What are the steps required to compile a fork of this repo?

Here are the steps I've taken, and perhaps someone can point out where I went wrong.
1. I deleted the Library Manager version of the library from my `~/Documents/Arduino/libraries` folder
2. I forked the repo to my own Github account [here](https://github.com/eparadis/Adafruit_Floppy)
3. I used `git clone` to clone my fork into `~/Documents/Arduino/libraries`.
4. I restarted my Arduino IDE (v2.3.4)
5. I opened the `01_floppy_capture_track_test` example using the `File > Examples > Adafruit_Floppy` menu item
6. I modified the example to use the Pico pinout. (I'm using a Pico 2)
7. I used the `Verify` button to compile the example

It seems to compile everything in the library and then fails to link with many lines referring to `multiple definitions` related to `Adafruit_BusIO`

At the end, it reports
```
Using library Adafruit Floppy at version 0.3.0 in folder: /Users/ed/Documents/Arduino/libraries/Adafruit_Floppy 
Using library Adafruit BusIO at version 1.17.0 in folder: /Users/ed/Documents/Arduino/libraries/Adafruit_BusIO 
Using library SPI at version 1.0 in folder: /Users/ed/Library/Arduino15/packages/rp2040/hardware/rp2040/4.4.2/libraries/SPI 
Using library SdFat at version 2.3.0 in folder: /Users/ed/Library/Arduino15/packages/rp2040/hardware/rp2040/4.4.2/libraries/SdFat 
Using library Wire at version 1.0 in folder: /Users/ed/Library/Arduino15/packages/rp2040/hardware/rp2040/4.4.2/libraries/Wire 
```

I have to admit I'm not very familiar with how modern Arduino libraries work. Is there some further step I need to take?

One thing that seems suspicious is that on the Release page for this project, it shows [v0.5.0](https://github.com/adafruit/Adafruit_Floppy/releases/tag/0.5.0) to be the latest Release. But that end report indicates I'm using `v0.3.0`. 

But the `library.properties` file [on the `main` branch also says it is 0.3.0](https://github.com/adafruit/Adafruit_Floppy/blob/main/library.properties) so perhaps this is just a file that needs updating and has nothing to do with my issue.

Thank you for taking the time to write this and read my issue!

jepler, 2025-01-23:
I don't know why you are seeing a build failure -- we test-compile many of the example sketches during our CI process, including 01_floppy_capture_track_test, on a couple of different boards, and the process doesn't encounter any errors.

You mentioned running on pico 2. We didn't test on any rp2350 boards during CI, but I have added a PR to test build on Feather RP2350 during CI (#50) -- this failed, though in an uninteresting way, because feather 2350 is not supported by our CI scripts yet. Locally, I can build the 01_floppy_capture_track_test example for feather rp2350 hstx and don't get any multiple definition errors.

I also entered a PR (#49) about the version number reporting and asked for help from a colleague. I do not release a lot of Arduino libraries and it looks like I forgot some manual steps required before a release.

To help continue to troubleshoot this, please show the modifications you made to the example, and your best guess at the relevant parts of the error messages during the build. Please use the code buttons or markdown syntax to make the text display properly.  Make sure you include the library version info too.

```
/home/jepler/.local/arduino-1.8.16/arduino-builder -compile -logger=machine -hardware /home/jepler/.local/arduino-1.8.16/hardware -hardware /home/jepler/.arduino15/packages -hardware /home/jepler/Arduino/hardware -tools /home/jepler/.local/arduino-1.8.16/tools-builder -tools /home/jepler/.local/arduino-1.8.16/hardware/tools/avr -tools /home/jepler/.arduino15/packages -built-in-libraries /home/jepler/.local/arduino-1.8.16/libraries -libraries /home/jepler/Arduino/libraries -fqbn=pico:rp2040:adafruit_feather_rp2350_hstx:flash=8388608_0,freq=150,opt=Small,rtti=Disabled,stackprotect=Disabled,exceptions=Disabled,dbgport=Disabled,dbglvl=None,usbstack=tinyusb,ipbtstack=ipv4only,boot2=boot2_w25q080_2_padded_checksum,uploadmethod=default -ide-version=10816 -build-path /tmp/arduino_build_965116 -warnings=none -build-cache /tmp/arduino_cache_362636 -prefs=build.warn_data_percentage=75 -verbose /home/jepler/src/Adafruit_Floppy/examples/01_floppy_capture_track_test/01_floppy_capture_track_test.ino
[...]
Using board 'adafruit_feather_rp2350_hstx' from platform in folder: /home/jepler/Arduino/hardware/pico/rp2040
Using core 'rp2040' from platform in folder: /home/jepler/Arduino/hardware/pico/rp2040
[...]
Using library Adafruit_Floppy at version 0.3.0 in folder: /home/jepler/Arduino/libraries/Adafruit_Floppy 
Using library Adafruit_BusIO at version 1.15.0 in folder: /home/jepler/Arduino/libraries/Adafruit_BusIO 
Using library SPI at version 1.0 in folder: /home/jepler/Arduino/hardware/pico/rp2040/libraries/SPI 
Using library SdFat at version 2.2.3 in folder: /home/jepler/Arduino/libraries/SdFat 
Using library Wire at version 1.0 in folder: /home/jepler/Arduino/hardware/pico/rp2040/libraries/Wire 
Using library Adafruit_TinyUSB_Library at version 3.1.3 in folder: /home/jepler/Arduino/libraries/Adafruit_TinyUSB_Library
[...]
Sketch uses 68164 bytes (0%) of program storage space. Maximum is 8380416 bytes.
Global variables use 163180 bytes (31%) of dynamic memory, leaving 361108 bytes for local variables. Maximum is 524288 bytes.
```

eparadis, 2025-01-23:
The modifications I made to the `01_floppy_capture_track_test` example are
1. comment out lines 3 to 34
2. comment out lines 51 to 55

The result is the pin definitions of:
```
#define DENSITY_PIN 2 // IDC 2
#define INDEX_PIN 3   // IDC 8
#define SELECT_PIN 4  // IDC 12
#define MOTOR_PIN 5   // IDC 16
#define DIR_PIN 6     // IDC 18
#define STEP_PIN 7    // IDC 20
#define WRDATA_PIN 8  // IDC 22 (not used during read)
#define WRGATE_PIN 9  // IDC 24 (not used during read)
#define TRK0_PIN 10   // IDC 26
#define PROT_PIN 11   // IDC 28
#define READ_PIN 12   // IDC 30
#define SIDE_PIN 13   // IDC 32
#define READY_PIN 14  // IDC 34
#ifndef USE_TINYUSB
#error "Please set Adafruit TinyUSB under Tools > USB Stack"
#endif
```

These are simply the original lines 35 through 50.

The version of the library I'm using is the `HEAD` of `main`
```
commit a8546adaf12b727c678ed21cc5506b2d40866745 (HEAD -> main, origin/main, origin/HEAD)
Merge: f898470 5fed247
Author: Jeff Epler <jepler@gmail.com>
Date:   Wed Jan 22 17:57:40 2025 -0600

    Merge pull request #47 from jepler/floppsy-build
    
    build mfm emulator during CI
```

(This currently still has the version set to 0.3.0, awaiting the merge of your PR to bump it to 0.6.1)

Here's the mess of output from the Arduino IDE:
```
Linking everything together...
/Users/ed/Library/Arduino15/packages/rp2040/tools/pqt-python3/1.0.1-base-3a57aed-1/python3 -I /Users/ed/Library/Arduino15/packages/rp2040/hardware/rp2040/4.4.2/tools/simplesub.py --input /Users/ed/Library/Arduino15/packages/rp2040/hardware/rp2040/4.4.2/lib/rp2350/memmap_default.ld --out /Users/ed/Library/Caches/arduino/sketches/D9B4422A4652A5E1A6AB7571B892BAF0/memmap_default.ld --sub __FLASH_LENGTH__ 4186112 --sub __EEPROM_START__ 272621568 --sub __FS_START__ 272621568 --sub __FS_END__ 272621568 --sub __RAM_LENGTH__ 512k --sub __PSRAM_LENGTH__ 0x000000
/Users/ed/Library/Arduino15/packages/rp2040/tools/pqt-gcc/4.0.1-8ec9d6f/bin/arm-none-eabi-gcc -Werror=return-type -Wno-psabi -DUSE_TINYUSB -I/Users/ed/Library/Arduino15/packages/rp2040/hardware/rp2040/4.4.2/libraries/Adafruit_TinyUSB_Arduino/src/arduino -DUSBD_PID=0x000f -DUSBD_VID=0x2e8a -DUSBD_MAX_POWER_MA=250 "-DUSB_MANUFACTURER=\"Raspberry Pi\"" "-DUSB_PRODUCT=\"Pico 2\"" -DLWIP_IPV6=0 -DLWIP_IPV4=1 -DLWIP_IGMP=1 -DLWIP_CHECKSUM_CTRL_PER_NETIF=1 -DFILE_COPY_CONSTRUCTOR_SELECT=FILE_COPY_CONSTRUCTOR_PUBLIC -DUSE_UTF8_LONG_NAMES=1 -DSDFAT_FILE_TYPE=3 -DDISABLE_FS_H_WARNING=1 "-DARDUINO_VARIANT=\"rpipico2\"" -DPICO_FLASH_SIZE_BYTES=4194304 @/Users/ed/Library/Arduino15/packages/rp2040/hardware/rp2040/4.4.2/lib/rp2350/platform_def.txt -mcpu=cortex-m33 -mthumb -march=armv8-m.main+fp+dsp -mfloat-abi=softfp -mcmse -ffunction-sections -fdata-sections -fno-exceptions -O3 -u _printf_float -u _scanf_float -c /Users/ed/Library/Arduino15/packages/rp2040/hardware/rp2040/4.4.2/boot2/rp2350/none.S -I/Users/ed/Library/Arduino15/packages/rp2040/hardware/rp2040/4.4.2/pico-sdk/src/rp2350/hardware_regs/include/ -I/Users/ed/Library/Arduino15/packages/rp2040/hardware/rp2040/4.4.2/pico-sdk/src/common/pico_binary_info/include -o /Users/ed/Library/Caches/arduino/sketches/D9B4422A4652A5E1A6AB7571B892BAF0/boot2.o
/Users/ed/Library/Arduino15/packages/rp2040/tools/pqt-gcc/4.0.1-8ec9d6f/bin/arm-none-eabi-g++ -L/Users/ed/Library/Caches/arduino/sketches/D9B4422A4652A5E1A6AB7571B892BAF0 -Werror=return-type -Wno-psabi -DUSE_TINYUSB -I/Users/ed/Library/Arduino15/packages/rp2040/hardware/rp2040/4.4.2/libraries/Adafruit_TinyUSB_Arduino/src/arduino -DUSBD_PID=0x000f -DUSBD_VID=0x2e8a -DUSBD_MAX_POWER_MA=250 "-DUSB_MANUFACTURER=\"Raspberry Pi\"" "-DUSB_PRODUCT=\"Pico 2\"" -DLWIP_IPV6=0 -DLWIP_IPV4=1 -DLWIP_IGMP=1 -DLWIP_CHECKSUM_CTRL_PER_NETIF=1 -DFILE_COPY_CONSTRUCTOR_SELECT=FILE_COPY_CONSTRUCTOR_PUBLIC -DUSE_UTF8_LONG_NAMES=1 -DSDFAT_FILE_TYPE=3 -DDISABLE_FS_H_WARNING=1 "-DARDUINO_VARIANT=\"rpipico2\"" -DPICO_FLASH_SIZE_BYTES=4194304 @/Users/ed/Library/Arduino15/packages/rp2040/hardware/rp2040/4.4.2/lib/rp2350/platform_def.txt -mcpu=cortex-m33 -mthumb -march=armv8-m.main+fp+dsp -mfloat-abi=softfp -mcmse -ffunction-sections -fdata-sections -fno-exceptions -O3 -u _printf_float -u _scanf_float @/Users/ed/Library/Arduino15/packages/rp2040/hardware/rp2040/4.4.2/lib/rp2350/platform_wrap.txt @/Users/ed/Library/Arduino15/packages/rp2040/hardware/rp2040/4.4.2/lib/core_wrap.txt -Wl,--cref -Wl,--check-sections -Wl,--gc-sections -Wl,--unresolved-symbols=report-all -Wl,--warn-common -Wl,--undefined=runtime_init_install_ram_vector_table -Wl,--undefined=__pre_init_runtime_init_clocks -Wl,--undefined=__pre_init_runtime_init_bootrom_reset -Wl,--undefined=__pre_init_runtime_init_early_resets -Wl,--undefined=__pre_init_runtime_init_usb_power_down -Wl,--undefined=__pre_init_runtime_init_clocks -Wl,--undefined=__pre_init_runtime_init_post_clock_resets -Wl,--undefined=__pre_init_runtime_init_spin_locks_reset -Wl,--undefined=__pre_init_runtime_init_boot_locks_reset -Wl,--undefined=__pre_init_runtime_init_bootrom_locking_enable -Wl,--undefined=__pre_init_runtime_init_mutex -Wl,--undefined=__pre_init_runtime_init_default_alarm_pool -Wl,--undefined=__pre_init_first_per_core_initializer -Wl,--undefined=__pre_init_runtime_init_per_core_bootrom_reset -Wl,--undefined=__pre_init_runtime_init_per_core_h3_irq_registers -Wl,--undefined=__pre_init_runtime_init_per_core_irq_priorities -Wl,--script=/Users/ed/Library/Caches/arduino/sketches/D9B4422A4652A5E1A6AB7571B892BAF0/memmap_default.ld -Wl,-Map,/Users/ed/Library/Caches/ard [... cut ...]
Using library Adafruit TinyUSB Library at version 3.4.2 in folder: /Users/ed/Library/Arduino15/packages/rp2040/hardware/rp2040/4.4.2/libraries/Adafruit_TinyUSB_Arduino 
exit status 1

Compilation error: exit status 1
```

The mentioned `SdFat_-_Adafruit_Fork` seems suspicious, but as I recall, that as installed by the IDE when I originally installed the library from the Library Manager. It mentioned it as a required library and installed it. It looks to come from https://github.com/adafruit/SdFat.git

eparadis, 2025-01-23:
I finally noticed that those file names in the BusIO library sure looked suspicious. Why all the "whatever 2.cpp" ? So I 
1. Deleted that directory in my `libraries` folder (just to make sure)
2. Uninstalled the "Adafruit BusIO" library from the IDE Library Manager
3. Re-Installed the "Adafruit BusIO" library (v1.17.0) from the IDE Library Manager

And now it compiles.  🤷 

I have no idea how I ended up with two over-layed copies of that library. I wonder if it has something to do with iCloud syncing my Documents folder? I recently added a new computer to my iCloud account that runs a different version of MacOS and perhaps it got confused somewhere as I switched working on this project in two different places.

Thanks for your assistance, and I look forward to digging into the code.

jepler, 2025-01-23:
[your message came in while I was writing this, I'll go ahead and post it]

This message seems to be saying that the two definitions each duplicate symbol come from a file called `Users/ed/Documents/Arduino/libraries/Adafruit_BusIO/Adafruit_SPIDevice.cpp` and one called `/Users/ed/Documents/Arduino/libraries/Adafruit_BusIO/Adafruit_SPIDevice 2.cpp`

I didn't check but I really doubt that Adafruit_BusIO has a file named `Adafruit_SPIDevice 2.cpp`.

Please make sure you can compile a busio SPIDevice example.

jepler, 2025-01-23:
I'm glad you got a successful build now!

closed 2025-01-23 by jepler