adafruit/Adafruit_Floppy #55: Issue setting up project "Failed to spin up motor & find index pulse" with Raspberry Pi Pico

TheMLGladiator, 2025-05-11:
I've been having trouble getting this project up and running with my Raspberry Pi Pico.
My Setup:
1. Clean install of Arduino IDE 2.3.6
2. Add link to philhower core in preferences
3. Install philhower core 4.5.3
4. Tools -> Set Pi Pico as board and USB stack: Adafruit TinyUSB
5. Install Adafruit Floppy 0.6.1 with all dependencies
6. Open example 01_floppy_capture_track_test and connect board to drive following the pin definitions:
`
#elif defined(ARDUINO_RASPBERRY_PI_PICO)
#define DENSITY_PIN 2 // IDC 2
#define INDEX_PIN 3   // IDC 8
#define SELECT_PIN 4  // IDC 12
#define MOTOR_PIN 5   // IDC 16
#define DIR_PIN 6     // IDC 18
#define STEP_PIN 7    // IDC 20
#define WRDATA_PIN 8  // IDC 22 (not used during read)
#define WRGATE_PIN 9  // IDC 24 (not used during read)
#define TRK0_PIN 10   // IDC 26
#define PROT_PIN 11   // IDC 28
#define READ_PIN 12   // IDC 30
#define SIDE_PIN 13   // IDC 32
#define READY_PIN 14  // IDC 34
`
7. Compile/Upload the project and open the serial monitor, setting baud to 115200

All I have been able to get is 
`
its time for a nice floppy transfer!
Sample freqency 24.00MHz
Waiting for index pulse...Didn't find an index pulse!
Failed to spin up motor & find index pulse
`
I have tried different versions of the floppy library and philhower core and two different floppy drives. Nothing seems to work. I've read some things about adding a pull up resistor between the READ_PIN and the microcontroller. My wiring could be wrong but that hasn't worked either.

ladyada, 2025-05-11:
hiya! we don't provide tech support for DIY builds but someone else might come along and help.

mbulandra, 2025-05-14:
Sorry for asking if it is obvious for you, but did you connect any GND pins between Pico and the floppy?

TheMLGladiator, 2025-05-15:
> Sorry for asking if it is obvious for you, but did you connect any GND pins between Pico and the floppy?

All good! I made sure to connect a few of the ground pins. Pin 1 was the main one I used when troubleshooting. The main drive I was using was a Samsung SFD-321B/LE. Just found this technical manual for the base model: https://jope.fi/drives/SAMSUNG-SFD321B-070103.pdf

ladyada-eagleclaw, 2026-10-12:
Closing this as a DIY wiring/setup support request, consistent with the support scope noted above. This thread has not established a reproducible library defect, so this closure does not mean the reported hardware problem was fixed. If a minimal reproducer identifies a library bug, please open a focused issue with the sketch, board/core/library versions, drive model, and measured signal behavior.

closed 2026-10-12 by ladyada-eagleclaw